gavinlyonsrepo/RpiMotorLib #1: Nema 17 stepper with DRV8825 only vibrating

williammlu, 2018-04-16:
I was looking to use this library with a stepper motor and the 8825 driver, but after running `A4998NemaTest.py`, the motor vibrates but does not step. I have been also running my own [script on the side](https://github.com/williammlu/toasty_bot/blob/master/scripts/double_rotate.py), which makes the stepper rotate (given that I run the pigpiod daemon).

Are there additional steps I need to do before getting this library to work with an 8825?

gavinlyonsrepo, 2018-04-16:
Hi 

Thank you for feedback 

As it says in Readme.md I have **_only_** tested it for a4988 
I am still waiting on a drv-8825 part to arrive. 
I mentioned drv-8825 in readme together with link. as pololu said here about drv-8825 at link
"Note that we carry several stepper motor drivers that can be used as alternatives for this module (and drop-in replacements in many applications):" and also "Module size, pinout, and interface match those of our A4988 stepper motor driver carriers in most respects"
https://www.pololu.com/product/1182

Now having said that I have spent sometime reading datasheet and noticed the truth table for micro resolution is  different for two components  from micro step 1/16 on-wards
https://www.pololu.com/product/2133
https://www.pololu.com/file/0J450/A4988.pdf
But code should STILL work for higher steps.

Can you post your code please?

Also double check that the coils of motor are connected to right pins of module although you said
you tested on an another library so this is probably not it. 
Vibrating stepper motor is a sign of motor coils not connected to right pins.

I will remove reference to DRv8825 from readme until I have fully tested component.

williammlu, 2018-04-18:
Sorry, I think this might have been a mistake on my part.

I used to running this [code](https://github.com/williammlu/toasty_bot/blob/master/scripts/double_rotate.py), which was adapted from [this tutorial](https://www.rototron.info/raspberry-pi-stepper-motor-tutorial/). For some reason, I was not able to get your code `A4998NemaTest.py` to get my stepper motor to rotate, but the `double_rotate.py` was working.

I ended up writing a [lot of similar code to better interface](https://github.com/williammlu/toasty_bot/blob/master/run/run.py) with my stepper motors, but then I realized only one of my two stepper motors was consistently working. The other one was frequently stopping, vibrating, and occasionally rotating. There was no pattern to the vibration versus rotation on that stepper, so I tried swapping leads and using other driver boards but the bad stepper was consistently failing.

After trying my good stepper motor with your test for the A4998, it ended up working well. It appears to be rotating appropriate 180 degrees, but I have not had the chance to check carefully. Also, I have only been using full stepping with your library and my scripts.

I'm not sure what is going on with my bad stepper motor, as it has inconsistent rotation. It has really puzzled me for the past couple of days, and I am assuming for now that it is damaged in some way.

gavinlyonsrepo, 2018-05-30:
Version 2.1-5 released fully supporting the DRV8825.

closed 2018-05-30 by gavinlyonsrepo