gavinlyonsrepo/RpiMotorLib #4: User Keyboard Interrupt : RpiMotorLib:

luxedo, 2019-05-21:
When I stop the execution of my script with `Ctrl+C`, occasionally I get this message:
```bash
User Keyboard Interrupt : RpiMotorLib:
```
But the code keeps running. I have to press a couple more times to stop it.

What's weird is that the behavior is not consistent, sometimes I won't get the message and the script stops. My guess is that when the motor is running I can't stop the script.

gavinlyonsrepo, 2019-05-22:
Hi 

Thank you for feedback and input.
I have tested this and the code is working as designed. 

1. Issue message "User Keyboard Interrupt : RpiMotorLib".
2. Print verbose report if  enabled.
3. Turn off GPIO pins related to motor.
4. Hand back control to script that called RpiMotorLib.

The keyboard interrupt is NOT designed to stop main program which called RpiMotorLib, but to stop motor and exit RpiMotorLib. I have been unable to replicate any circumstance where motor kept going.
I am unable to merge this Pull request as i don't want program to halt on Ctrl+C its up to main program 
to stop not the library.

Can I see main program/test code that you use to call RpiMotorLib?
What motor are you running?

While testing , I did notice one anomaly which could cause inconsistent behaviour.
If a user executed Ctrl+c before Motor run started for example  during init delay, 
The verbose print report at end would cause another exception in the exception routine. 
As a variable was not assigned yet:.
**_"UnboundLocalError: local variable 'step_sequence' referenced before assignment"_**

I order to fix this I create a new version. where the verbose report is in the "else" code block of exception handler rather than the "finally". I.e it will only be executed if the motor finished its run with no exceptions. Version 2.6-7 is now ready.

Regards

luxedo, 2019-05-23:
I see. Indeed it's working as you described. I'm running a couple of 28BYJ motors.

I had issues with this because my use case makes several motor adjustments in small intervals of time, so it is hard to stop at the main script.

Thanks for the feedback

closed 2019-05-23 by luxedo

DavidsPaics, 2024-05-17:
Hi.
I just want to understand if there is a way to easily remove this weird feature in the main program using this library.

It is really annoying and bad practice to prevent any keyboard interrupts. This should really be a flag or parameter you can set when initializing a motor that defaults to off.

luxedo, 2024-05-18:
This could at least be an optional feature and leave for the user to decide whether he wants to interrupt the program or just the function:

```python
RpiMotorLib.captureSigint(False)
```

DavidsPaics, 2024-05-18:
#5 
I suggest merging the pull request that already fixes this issue. Preventing keyboard interrupts is only useful in very specific cases when developing this library and is always harmful to the library’s user.