gavinlyonsrepo/RpiMotorLib #6: i used 103H5210-0457 dvr8825 but strange...

MAKER-park, 2020-02-20:
hello i try this example
import RPi.GPIO as GPIO

# import the library
from RpiMotorLib import RpiMotorLib
    
#define GPIO pins
GPIO_pins = (14, 15, 18) # Microstep Resolution MS1-MS3 -> GPIO Pin
direction= 20       # Direction -> GPIO Pin
step = 21      # Step -> GPIO Pin

# Declare a instance of class pass GPIO pins numbers and the motor type
mymotortest = RpiMotorLib.A4988Nema(direction, step, GPIO_pins, "DRV8825")

# call the function, pass the arguments
mymotortest.motor_go(False, "Full" , 100, .01, False, .05)

# good practise to cleanup GPIO at some point before exit
GPIO.cleanup()

and my motor 103H5210-0457 this is just use 24v ... so i just use and adjust current power...

but 
It's trembling, it's vibrating ... And it's feverish ... Can you tell me how to match it in my code?

gavinlyonsrepo, 2020-02-20:
Hi 

This is the classic failure mode of a wrongly connect stepper motor. 

I am having trouble finding a datasheet for the 103H5210-0457 motor stepper.

**Is it unipolar or bipolar?**

https://www.ebay.com/itm/Used-SANYO-DENKI-103H5210-0457-StepSSyn-STEPPING-MOTOR-1-2A-/113357276946

The pictures I see on ebay etc show a six wire motor, if this  is what you have this is  a unipolar motor. This DRV8825 and this code class  is for bipolar motor ONLY. 
You cannot run a unipolar with it.
 
There is a "BYJMotor"  class for controlling unipolar motors but you need a additional components.
to control.  

regards

gavinlyonsrepo, 2020-02-26:
Closed ,   not a software issue.

closed 2020-02-26 by gavinlyonsrepo