gavinlyonsrepo/RpiMotorLib #7: The DC motor slows down after some time for forward and backward movement

Bis-bots, 2020-10-01:
The driver is moving the motors but if the movement is continued the motor slows down and stops for a while (1 sec). The first time it happens just after around 5 secs of running the motor. Then the motor starts running again and it happens repeatedly after about a minute. My guess is there is some sleep time set in the library. 
Related code I am using in raspberry pi 3B is

import time
import RPi.GPIO as GPIO
import pigpio
#from RpiMotorLib import RpiMotorLib
from time import sleep
from RpiMotorLib import rpi_pservo_lib
from RpiMotorLib import rpi_dc_lib
MotorOne = rpi_dc_lib.L298NMDc(6 ,12 ,13 ,30 ,False, "motor_one")
MotorTwo = rpi_dc_lib.L298NMDc(26 ,20 ,21 ,30 ,False, "motor_two")

        MotorOne.forward(30)
        MotorTwo.forward(30)

gavinlyonsrepo, 2020-10-01:
HI 

Try 

   
 MotorOne = rpi_dc_lib.L298NMDc(GPIO , GPIO , GPIO , freq , verbose, name)

MotorOne = rpi_dc_lib.L298NMDc(6 ,12 ,13 ,30 ,False, "motor_one")
MotorTwo = rpi_dc_lib.L298NMDc(26 ,20 ,21 ,30 ,False, "motor_two")

change to 

MotorOne = rpi_dc_lib.L298NMDc(6 ,12 ,13 ,50 ,False, "motor_one")
MotorTwo = rpi_dc_lib.L298NMDc(26 ,20 ,21 ,50 ,False, "motor_two")

i.e Set the freq setting to 50 hz not 30 and try again, this is what is in  documentation and test code.

regards.

Bis-bots, 2020-10-02:
Hi

Thanks fro your quick reply.
Actually I tried the following declaration first.
MotorOne = rpi_dc_lib.L298NMDc(6 ,12 ,13 ,50 ,False, "motor_one")
MotorTwo = rpi_dc_lib.L298NMDc(26 ,20 ,21 ,50,False, "motor_two")

But thought that the problem might come from the frequency difference in the declaration and in the main run. i.e.
    MotorOne.forward(30)
    MotorTwo.forward(30)
So I changed the frequency in the definition and made them similar to the main run.

Anyway,  
MotorOne = rpi_dc_lib.L298NMDc(6 ,12 ,13 ,50 ,False, "motor_one")
MotorTwo = rpi_dc_lib.L298NMDc(26 ,20 ,21 ,50,False, "motor_two")
did not resolve the problem.

But from some other sources I got some hints and declared the motors in the following way.
This resolved my problem. The motors are now running without any stop.
MotorOne = rpi_dc_lib.L298NMDc(6 ,12 ,13 ,1000 ,False, "motor_one")
MotorTwo = rpi_dc_lib.L298NMDc(26 ,20 ,21 ,1000,False, "motor_two")

Thanks anyway.

closed 2020-10-02 by Bis-bots