gavinlyonsrepo/RpiMotorLib #9: Issues with 28BYJ-48 motor

spodhuturi16, 2021-04-26:
I'm trying to run the example documentation but I keep getting a syntax error on line 281:

```
  File "libstepper.py", line 2, in <module>
    from RpiMotorLib import RpiMotorLib
  File "/usr/local/lib/python2.7/dist-packages/RpiMotorLib/RpiMotorLib.py", line 281
    print("Steps count {}".format(i+1), end="\r", flush=True)
                                           ^
SyntaxError: invalid syntax
```

Any help would be appreciated!

gavinlyonsrepo, 2021-04-26:
Hi 

I suspect you are running python2 instead of python3 

what is full  command are you running?

spodhuturi16, 2021-04-26:
Ahhh thank you that solved my issue! In the documentation the install line used pip not pip3 so I believe I thought the library was meant to be run with python, not python3. I reinstalled with pip3 and ran with python3 and it's working great! Thanks for the awesome library and quick response!

closed 2021-04-26 by spodhuturi16