gavinlyonsrepo/RpiMotorLib #11: Is there a way to run two nema stepper motors at the same time?

Choikyungho9, 2021-07-14:
Is there a way to run two nema stepper motors at the same time?
I'm trying to run two Nema stepper motors, not BYJ motors.  
the BYJ motor has a code that turns two at the same time. Is it impossible to run two Nema motors at the same time?

gavinlyonsrepo, 2021-07-15:
Hi 

Yes , in theory although I did not test it. Which  Motor controller are you using?.

Choikyungho9, 2021-07-15:
Hello. I'm using the Nema17 motor. Do you have any plans to test it?

gavinlyonsrepo, 2021-07-15:
Yes 

Nema 17 is motor type number, I  need the motor controller number (A3967 or A4988 or DRv8825 etc)
In order to write and test an example. 

regards

Choikyungho9, 2021-07-15:
oh that's good news
 
I think I gave you the wrong information. I am using the drv8825 driver.

gavinlyonsrepo, 2021-07-15:
hi 

Bad news.

I created a new test file based on test/MultiMotorThreading.py 
 test/MultiMotorThreading.py which works for BYJMotor.
The new file test file for DRV8825 is here at gist link :  https://gist.github.com/gavinlyonsrepo
It does NOT work.
The threads run one after another , instead of in parallel.
I don't know why.

I will have to investigate this when I have time.

Regards

Choikyungho9, 2021-07-16:
Thanks for your hard work.

When I run the code you tested, the motors do not run at the same time.

But I am very satisfied with this. Because with the code you provided, I will continue to find a way.

Thank you very much.

gavinlyonsrepo, 2021-07-17:
Hi 

Good news
I got it working,

 There are now two working basic setup files in the test section which show who to use threading to turn two motors in parallel. 

1. For Unipolar 28BYJ-48  MultiMotorThreading_BYJ.py
2. For Bipolar DRV8825 Stepper MultiMotorThreading_DRV8825.py

Regards

Choikyungho9, 2021-07-17:
Oh my gosh it works fine.

I really appreciate your efforts.

closed 2021-07-18 by gavinlyonsrepo