gavinlyonsrepo/RpiMotorLib #14: Gecko G251X

Zstack836, 2021-11-02:
Hello!

Looking to see if someone more knowledgeable then myself has any insight into if this library would be useful in controlling a NEMA17 Stepper (https://openbuildspartstore.com/nema-17-stepper-motor/) with a Gecko G251X stepper driver (https://www.geckodrive.com/g251x-digital-stepper-drive.html).

Per the documentation for the G251X, I should only need a step and direction input from the pi, which pretty much any of the different components offered in this library look like they do. I don't need to worry about any of the micro stepping pins so I could set those to 0 (or some other unused pin). 

Would anyone be able to provide their insight into if this would work or not? Am I missing something? Thanks in advance!

gavinlyonsrepo, 2021-11-03:
Hi 

I think we can hack/setup the A4988 test file/code for this.

As per the A4988 example file  A4988_Nema_Test.py  in test folder.
 
# Microstep Resolution MS1-MS3 -> GPIO Pin , can be set to (-1,-1,-1) to turn off resolution GPIO
    GPIO_pins = (-1, -1, -1)   

Next call using Full  steptype 

mymotortest.motor_go(False, "Full" , 200, .05, False, .05)

The only issue now is the verbose reporting of the motor will be incorrect as the G251X uses 1/10 step resolution as per user guide. The motor will turn as you order it but if verbose reporting is set to true  you will get wrong numbers for size of turn.  A simple hack to one line of  code can correct this, is this is issue.  

regards

Zstack836, 2021-11-03:
Thank you for the input!

I was able to get the stepper to move using your above recommendations with one minor change.

I had to change the **stepdelay** to something quite a bit smaller for it to move. Using the provided step delay of .05 the motor just seemed to pulse but not go anywhere. I dropped it down to .00001 at the smallest (tried going smaller and while the stepper did seem to go faster, it was not "stable" - was very jittery).

It still seems a bit "rough" at the start and stop, but I will keep playing around with it and see if I can figure anything else out. Any insight you may be able to provide would be appreciated! Thanks!

Godtec, 2021-11-27:
Hi Zstack:

There is an adjustment pot on the inside of the G251X unit. Did you try adjusting that?
The manual is here, Just wondering if you were able to get it?

[https://www.geckodrive.com/g251x-digital-stepper-drive.html](url)

Also, you can refer to LinuxCNC for the timing setup's of similar stepper motor drivers to compare timings.
At this point you will have to play to get it right. This page hopefully will get you started.

[http://wiki.linuxcnc.org/cgi-bin/wiki.pl?Stepper_Drive_Timing](url)

Thanks!
MikeK

Zstack836, 2021-11-28:
I had previously tuned the G251X unit per the user guide, but thanks for mentioning it just in case.

I played around with the numbers a lot more after the post update and was able to get it to move reliably, just not as fast as I would have liked. 

Ended up switching to a Teensy 3.2 dedicated controller to interface with the stepper. Dedicated hardware vs shared resources.

closed 2021-11-29 by gavinlyonsrepo