gavinlyonsrepo/RpiMotorLib #15: two steppers simultaneously example issue

kthod861, 2021-11-28:
Hello,

I was trying to use RpiMotorLib to manage 2 steppers at the same time (with different motions) but failed to do so.
I mean it work, but both steppers are playing one after the other, not simultaneously.

Config ; raspi4 > 2X easy drivers > 2X Nema 17.

here's the script i'm using based on your example. 
Am i missing something ?

Regards

`

"""
Test example file for module:rpiMotorlib.py
File: RpiMotorLib.py DRV8825 with A4988Nema class,
Desc:
Use threading to run two motors at same time.
use push button(to VCC) on GPIO 17 to stop motors if necessary
"""

import time
import concurrent.futures
import RPi.GPIO as GPIO

# Next 3 lines for development local library testing import
# Comment out in production release and change RpiMotorLib.A4988Nema to A4988Nema
#import sys
#sys.path.insert(0, '/home/pi/Documents/tech/RpiMotorLib/RpiMotorLib')
#from RpiMotorLib import A4988Nema

# Production installed library import
from RpiMotorLib import RpiMotorLib

# Motor 1
GPIO_pins_1 = (-1,-1)
direction_1 = 24  # Direction -> GPIO Pin
step_1  = 23     # Step -> GPIO Pin

# Motor 2
GPIO_pins_2 = (-1,-1)
direction_2 = 8  # Direction -> GPIO Pin
step_2  = 25      # Step -> GPIO Pin

# Declare two named instance of class, one for each motor
mymotortestOne = RpiMotorLib.A3967EasyNema(direction_1, step_1, GPIO_pins_1)
mymotortestTwo = RpiMotorLib.A3967EasyNema(direction_2, step_2, GPIO_pins_2)

def main():
    """main function loop"""

    # ====== tests for two motor DRV8835 ====
    # motor_go(clockwise, steptype, steps, stepdelay, verbose, initdelay)
    with concurrent.futures.ThreadPoolExecutor() as executor:
        f1 = executor.submit(mymotortestOne.motor_move(.0003, 1600*2 , False, True, "Full", .05))
        f2 = executor.submit(mymotortestTwo.motor_move(.0003*2.0, 1600 , False, True, "Full", .05))

# ===================MAIN===============================

if __name__ == '__main__':

    print("START")
    main()
    GPIO.cleanup()
    print("END")
    exit()

# =====================END===============================`

gavinlyonsrepo, 2021-11-28:
Hi 

I think you have a syntax error in these two lines 

f1 = executor.submit(mymotortestOne.motor_move(.0003, 16002 , False, True, "Full", .05))
f2 = executor.submit(mymotortestTwo.motor_move(.00032.0, 1600 , False, True, "Full", .05))

You change to 

		f1 = executor.submit(mymotortestOne.motor_move, .0003, 1600, False, True, "Full", .05)
		f2 = executor.submit(mymotortestTwo.motor_move, .0003, 1600 , False, True, "Full", .05)

the function name is taken as first parameter of executor.submit followed by the  functions parameters.

regards

kthod861, 2021-11-28:
Ho wow, so sorry to have annoyed you with that kind of a dumb mistake.

thanks a lot !

closed 2021-12-01 by gavinlyonsrepo