gavinlyonsrepo/RpiMotorLib #17: Can I run two stepper motors with L298n driver using the "MultiMotorThreading_DRV8825.py"?

zeph01, 2022-03-28:
Hi 

Good news
I got it working,

 There are now two working basic setup files in the test section which show who to use threading to turn two motors in parallel. 

1. For Unipolar 28BYJ-48  MultiMotorThreading_BYJ.py
2. For Bipolar DRV8825 Stepper MultiMotorThreading_DRV8825.py

Regards

_Originally posted by @gavinlyonsrepo in https://github.com/gavinlyonsrepo/RpiMotorLib/issues/11#issuecomment-881880460_

zeph01, 2022-03-28:
Thank you for responding!
I forgot to mention. I am using bipolar stepper motors. I need them both to move in one direction together, stop together, and then move in opposite direction together. 

I have not used concurrent.futures module before. 
Would this code work?

`state = 0
with concurrent.futures.ThreadPoolExecutor() as executor:
       while (True):
               if (state == 0):
                    f1 = executor.submit(mymotortestOne.motor_run, GpioPins1,.05,128, False, False, "half", .05)
                    f2 = executor.submit(mymotortestTwo.motor_run, GpioPins2,.05,128, False, False, "half", .05)
                    state = 1
               if (state == 1):
                   mymotortestOne.motor_stop()
                   mymotortestTwo.motor_stop()
                   state = 2
               if (state == 2):
                   f1 = executor.submit(mymotortestOne.motor_run, GpioPins1,.05,128, True, False, "half", .05)
                   f2 = executor.submit(mymotortestTwo.motor_run, GpioPins2,.05,128, True, False, "half", .05)
                   state = 3
               if (state == 3):
                   mymotortestOne.motor_stop()
                   mymotortestTwo.motor_stop()
                   state = 0`

gavinlyonsrepo, 2022-03-28:
Hi 

No that won't work. 
You can try it like this at gist link

https://gist.github.com/gavinlyonsrepo/a035f9c22a37d3dcf35aeb9a9006b2db

I have not tested this on PI as i am away now. But i think it will work.

regards

closed 2022-04-03 by gavinlyonsrepo