gavinlyonsrepo/RpiMotorLib #18: Stepper motor runs for ever

BadOwlEule, 2022-05-07:
Hi, I am currently trying to drive a Nema 17 stepper motor with my Raspberry Pi using an A4988 Stepper Motor Driver. When I run the python scripts they work so far, but when the script is terminated with GPIO.cleanup() the motor slowly rotates forever in one direction. I am quite new to this, what am I doing wrong?
Best regards
Valentin

gavinlyonsrepo, 2022-05-07:
Hi

I would have to see your complete software and hardware setup.

BadOwlEule, 2022-05-07:
I have set up everything exactly as described in the documentation for Nema11 A4988 and have so far only executed the script for the motor test. What I just noticed is that I ordered an A4988 stepper driver and it says so on the package, but on the microprocessor it says HR4988. Does that make any difference?

BadOwlEule, 2022-05-07:
I am a little closer to solving the problem. GPIO.cleanup() sets all ports back to input, I think the stepper driver can't handle that.

gavinlyonsrepo, 2022-05-08:
Hi 

I have never heard of the HR4988 until now.  By doing research
The HR4988 appears to be not  100% the same as A4988 
Its a Chinese clone chip not an Allegro microsystems  product(A4988). It appears to have a bad reputation for noise and vibration going from some posts I have found. It is designed to operate bipolar stepper motors in full-,1/2-, 1/4-, 1/8-, 1/16-,1/32-,1/64- and 1/128- step modes, with an output drive. The A4988 only goes to 1/16.

I still don't know why it would not stop?
Are all grounds connected to each other ?
Is the power supply filtered by a capacitor?
Do you only have one part?
Have you tried at different microsteps settings?

gavinlyonsrepo, 2022-05-29:
Hi 

I am NOT seeing this on my setup which is ::

'0.7.0' and '0.7.1' RPi.GPIO ( I tested both 0.7.1 is latest PyPi feb 2022)
OS: Raspbian 10 buster
Kernel: armv7l Linux 5.10.103-v7+
Python 3.7.3 
Raspberry pi model 3B,

I need to get a more modern Pi at some point in near future, to investigate fully.
 I cannot see  any relevant modern issue ticket on the[ RPI.GPIO library, website](https://sourceforge.net/projects/raspberry-gpio-python/). 

Is it possible that people seeing the issue have an old version of RPi.GPIO installed?

gavinlyonsrepo, 2022-09-15:
Hi 

I have added "Note 2" to the readme to inform users, I will leave it in test scripts for the moment.

Regards

closed 2022-09-17 by gavinlyonsrepo