gavinlyonsrepo/RpiMotorLib #19: [question] motor turning back and forth

wsw70, 2022-08-18:
I am trying to rotate a stepper motor using `RpiMotorLib`, following the instructions of a [post in makersportal.com][2] (using the same equipment - a DVR8852 module and a [Nema 17](https://www.omc-stepperonline.com/nema-17-bipolar-45ncm-63-74oz-in-1-5a-42x42x39mm-4-wires-w-1m-pin-connector-17hs15-1504s-x1) and the initialization of the motor from the code in the article is

```python
mymotortest = RpiMotorLib.A4988Nema(direction, step, (21,21,21), "DRV8825")
```

The motor starts but moves back and forth. I reduced the step delay to `0.05` and I can see that **each step is in a different direction** (effectively not turning the motor)

I tried to understand what the tuple `(21,21,21)` means, the docs mention 

> `mode_pins` `type`=tuple of 3 ints, `help`=GPIO pins connected to, Microstep Resolution pins MS1-MS3 of IC. Pass in `(-1, -1, -1)` for hardwired MS-X connection

I do not understand that entry and passing `(-1,-1,-1)` does not change the behaviour.

Would you have an idea what the problem can be?

  [1]: https://github.com/gavinlyonsrepo/RpiMotorLib
  [2]: https://makersportal.com/blog/raspberry-pi-stepper-motor-control-with-nema-17

gavinlyonsrepo, 2022-08-18:
Hi 

That example is dated 01-2021,  I changed the code in 05-2021 in include the (-1,-1,-1) option
Setting the  (-1,-1,-1) is an option to hard-wire the logic MS-X resolution pins on some stepper motors where applicable to save GPIO pins. In the old code you could set them all the same to reduce GPIO from 3 to  1 that's why they(linked example) are using (21,21,21) 

Connect 5 GPIO pins to MS1, MS2, MS3, STEP and DIR. if you do not wish to use GPIO for MS_X pins and hard wire MS-X to logic levels as per Microstep Resolution Truth Table. You can also pass in (-1, -1, -1) to software. Do this if your project only uses one type of resolution and you wish to save GPIO pins. If you do this you must pass correct resolution to software to get correct results.

https://github.com/gavinlyonsrepo/RpiMotorLib/blob/master/Documentation/Nema11A4988.md

Anyway, back and forth motion or Vibrating stepper motor is a sign of motor coils not connected to right pins, most  likely.
Can you verify you have identified correct the two coils and connected them to correct outputs as per diagram.

https://github.com/gavinlyonsrepo/RpiMotorLib/blob/master/images/wiringdiagram1.jpg

Regards

wsw70, 2022-08-18:
Thank you very much, I swapped the pins and it is OK now. The schema for the article I linked to is incorrect.

closed 2022-08-18 by wsw70