gavinlyonsrepo/RpiMotorLib #20: Turn -90°

Aterox-os, 2022-10-05:
Hello I´m using this library to controll a 28BYJ stepper motor. I want to be able to enter an angle and then the motor turn to that position. I wrote the following code:

```
from RpiMotorLib import RpiMotorLib
import RPi.GPIO as GPIO

GpioPins = [21,20,16,12]

val = int(input("val:"))
stepsByAngle = round((val*0.71111111)*2)

mymotortest = RpiMotorLib.BYJMotor("MyMotorOne", "28BYJ")

if stepsByAngle > 0:
        mymotortest.motor_run(GpioPins , .002, stepsByAngle, False, True, "half", .05)
        print("positive")
elif stepsByAngle < 0:
        mymotortest.motor_run(GpioPins , .002, stepsByAngle, True, True, "half", .05)
        print("negative")

GPIO.cleanup()
```
It´s working properly. I enter an angle and the code translates it into steps and the motor turn exactly 90°. I want also that the motor can turn -90° but that doesn´t work. What´s the problem?

gavinlyonsrepo, 2022-10-05:
Hi
 
What do you mean by "that doesn´t work".

So you entering -90 at command line I assume.
That means "stepsByAngle"  will be -128  which from a code point of view is 0
So in the "negative branch" of the if loop you need to convert "stepsByAngle" to positive.
You can use the abs() function for this or simply multiply by -1.

```
elif stepsByAngle < 0:
        stepsByAngle =  stepsByAngle * -1  # makes -128 = 128
        mymotortest.motor_run(GpioPins , .002, stepsByAngle, True, True, "half", .05)
        print("negative")
```

Regards

Aterox-os, 2022-10-05:
Thanks for the quick answer. You´re right I didn´t though about that. Now it works.

gavinlyonsrepo, 2022-10-07:
I will put in error checking for negative step numbers in the next version maybe. closing issue.

closed 2022-10-07 by gavinlyonsrepo