gavinlyonsrepo/RpiMotorLib #21: L298n Motor problem

Antonstr, 2022-10-14:
Good day to you

Trying to run the following code in Python

from time import sleep
from RpiMotorLib.rpi_dc_lib import L298NMDc

left_motor = L298NMDc(6, 13, 18, name="left")
left_motor.forward()
sleep(5)
left_motor.stop()
left_motor.cleanup(True)

# Change pin order
left_motor = L298NMDc(13, 6, 18, name="left")
left_motor.forward()
sleep(5)
left_motor.stop()
left_motor.cleanup(True)

It returns the following

pi@raspberrypi:~ $ python3 motorl.py
Traceback (most recent call last):
  File "motorl.py", line 11, in <module>
    left_motor = L298NMDc(13, 6, 18, name="left")
  File "/usr/local/lib/python3.7/dist-packages/RpiMotorLib/rpi_dc_lib.py", line 58, in __init__
    self.my_pwm = GPIO.PWM(self.pwm_pin, self.freq)
RuntimeError: A PWM object already exists for this GPIO channel
pi@raspberrypi:~ $

Could you please assist

Thank you sincerely

Anton

bessman, 2022-10-14:
To clarify, Anton and I were trying to figure out in which order the pins should go to make the motor go forward. But it seems like the PWM pin does not get reset properly when calling the `cleanup` method.

gavinlyonsrepo, 2022-10-15:
Hi 

I was able to replicate this problem here. 
RpiMotorLib "cleanup" method calls the RPIO.GPIO.cleanup method(if "true" is passed as argument). So that's  where the issue is not clearing the PWM channel. Issue is with  RPIO.GPIO.cleanup upstream. 
This was being reported before see [issue 18](https://github.com/gavinlyonsrepo/RpiMotorLib/issues/18)
At the time I was not able to replicate issue so I just added a note to readme see notes section.

Usually "cleanup" would be called at end of program not in the middle as you doing. As a workaround you can clear the GPIO manually or simply delete the object like so using the "del" command.
I have just tested in the terminal , not with the motor controller but it seems to work , no errors

`from time import sleep
from RpiMotorLib.rpi_dc_lib import L298NMDc

left_motor = L298NMDc(6, 13, 18, name="left")
left_motor.forward()
sleep(5)
left_motor.stop()
left_motor.cleanup(True)

del left_motor # Added , deletes object 

# Change pin order
left_motor = L298NMDc(13, 6, 18, name="left")
left_motor.forward()
sleep(5)
left_motor.stop()
left_motor.cleanup(True)`  

Can you tell me ? 
What Raspberry PI you use
Kernal version ?
What RPi.GPIO version ?
What Operating system?

regards

gavinlyonsrepo, 2022-10-15:
Hi 

I have discovered I can stop the PWM  by adding "self.my_pwm.stop()" to the cleanup method
, so I do not have to depend on whatever has gone wrong with the RPI.GPIO.cleanup method.
I will add this in in the next version. 

regards

Antonstr, 2022-10-15:
Hi Gavin

I changed the GPIO pins to your configuration and all is well now
I must just say that the motor control is more than noticeably smoother
than before

Thank you

Sincerely

Antonstr, 2022-10-15:
Can you tell me ?
What Raspberry PI you use Rpi 3b
Kernal version ? 5..1.103-v7+
What RPi.GPIO version ? 0.7.0.egg-info
pigpio version? 1.78.egg-info
What Operating system? Raspian 10 Buster

gavinlyonsrepo, 2022-10-16:
Hi 

New version  3.2-3 released.
I have added the pwm.stop command to the cleanup method in  all DC motor classes.

Regards

closed 2022-10-16 by gavinlyonsrepo

Antonstr, 2022-10-16:
Thank you
Appreciated