gavinlyonsrepo/RpiMotorLib #23: Motor stops working after running Test Code

philipprsch, 2023-09-08:
Hello, what I did was simply run this code: test/L298_Step_Test.py, when after less than two seconds, during which the motor made some light clicking sounds (as if it was trying to rotate), it suddenly stopped working. I dont know wether this problem relates to the Raspberry Pi, Stepper Motor or the L298 Driver Board I am using. 
Here are some observations I did after it happend with my multimeter:
The GPIOS are capable of outputting current.
The Driver Board stepper motor pins show a voltage of 1-2V even when I am not executing any code.
The Driver Board red led is on.
The Driver board is not heating as before, when the motor was still turning.
I assume the stepper motor coils are still intact, as I probed the blue/red and green/black wires and in both cases it produced a beeping sound.

I would appricate any suggestions as to what the problem might be.

gavinlyonsrepo, 2023-09-08:
Hi 

"When after less than two seconds, during which the motor made some light clicking sounds (as if it was trying to rotate)"

This problem sounds like a vibrating motor .  Vibrating stepper motor is a sign of motor coils not connected to right pins. 

From the documentation page.,
[image]

https://github.com/gavinlyonsrepo/RpiMotorLib/blob/master/Documentation/Nema11L298N.md

Can  you double check this ?
Also 
What Power supply are you using?
What stepper motor are you using?

Regards

philipprsch, 2023-09-09:
Hello,
i rechecked the connection as mentioed on the documentation page.
I am using a 12V 6A high quality power supply with a (bipolar) Nema 17 stepper motor.

After the light vibration just after running the code, now the motor doesent move at all and I cant hear anything that woud suggest that there is current flowing through the coils, so I am asuming the Driver was burned out.

However I cant  understand why this would only happen after running this particular code (could also have been a coincidence), as I did not have any problems before, when using my own code:

import RPi.GPIO as GPIO
import time
import math
import asyncio

def circular(min, max, val):
    if val > max : return min
    elif val < min: return max
    else: return val

class Stepper:
    def __init__(self, gpios, mode="half", sleep=0.01, max_half_steps=4096):
        self.modes = {
          "half": [
              [0],[0,1],[1],[1,2],[2],[2,3],[3],[3,0]
          ],
          "full": [
              [0,1],[1,2],[2,3],[3,0]
          ],
          "wave": [
              [0],[1],[2],[3]
          ]
        }
        self.max_half_stpes = max_half_steps
        self.mode = mode
        self.gpios = gpios
        self.seq_count = 0
        self.sleep = sleep
        self.position = 0

        self.full_rotation = self.max_half_stpes / (8 / len(self.modes[self.mode]))
        
        GPIO.setmode(GPIO.BCM)

        for gpio in self.gpios:
            GPIO.setup(gpio, GPIO.OUT)
            GPIO.output(gpio, GPIO.LOW)
    def __enter__(self):
        return self
    
    def __exit__(self, exc_type, exc_value, trace):
        print("Exiting")
        GPIO.cleanup()

    async def deg(self, deg):
        return round((deg/360) * self.full_rotation)
    
    async def rotate(self, steps, dir=1):
        for i in range(abs(steps)):
            for pin in range(len(self.gpios)):
                if (pin in self.modes[self.mode][self.seq_count]):
                    GPIO.output(self.gpios[pin], GPIO.LOW)
                else:
                    GPIO.output(self.gpios[pin], GPIO.HIGH)
            self.position += circular(0, self.full_rotation-1, self.position+dir)
            await asyncio.sleep(self.sleep)
            self.seq_count = circular(0, len(self.modes[self.mode])-1, self.seq_count+dir)

async def rotator():
    await stepper.rotate(await stepper.deg(90), -1) #Bring sensors in start position
    dir = 1 #Counterclockwise
    while True:
        await stepper.rotate(await stepper.deg(180) , dir)
        dir = dir*(-1)

with Stepper([13,6,5,0], mode="full", max_half_steps=200, sleep=0.002) as stepper:
    loop = asyncio.get_event_loop()
    loop.create_task(rotator())
    loop.run_forever()