gavinlyonsrepo/RpiMotorLib #25: Motor stop button don't work

umsmin, 2024-01-29:
When I comment the Stop button in like in the demo for A4988 then the motor don't stop. Only the following error on console:
`Test file: Stopping motor
Traceback (most recent call last):
  File "/usr/src/RpiMotorLib-3.2/test/./A4988_Nema_Test.py", line 90, in button_callback
    mymotortest.motor_stop()
NameError: name 'mymotortest' is not defined`

What have I made wrong?

Thanks

gavinlyonsrepo, 2024-01-29:
Hi 

Can you copy your Entire working test file here.

regards

umsmin, 2024-01-29:
`""" test example file for rpiMotorlib.py A4988 NEMA"""

import time
import RPi.GPIO as GPIO

"""
# Next 3 lines for development, local library testing import
# Comment out in production release and change RpiMotorLib.A4988Nema to A4988Nema
import sys
sys.path.insert(0, '/home/pi/Documents/tech/RpiMotorLib/RpiMotorLib')
from RpiMotorLib import A4988Nema
"""

# Production installed library import
from RpiMotorLib import RpiMotorLib

# Comment in To Test motor stop, put a push button to VCC on GPIO 17
GPIO.setwarnings(False)
GPIO.setmode(GPIO.BCM)
GPIO.setup(4, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)

def main():
    """main function loop"""

    # Comment in To Test motor stop , put push button to VCC on GPIO 17
    GPIO.add_event_detect(4, GPIO.RISING, callback=button_callback)

    # ====== Tests for motor ======

    # Microstep Resolution MS1-MS3 -> GPIO Pin , can be set to (-1,-1,-1) to turn off
    GPIO_pins = (14, 15, 18)
    direction= 21       # Direction -> GPIO Pin
    step = 20      # Step -> GPIO Pin

    # Declare an named instance of class pass GPIO-PINs
    # (self, direction_pin, step_pin, mode_pins , motor_type):
    mymotortest = RpiMotorLib.A4988Nema(direction, step, GPIO_pins, "A4988")

    # ====================== section A ===================
#    print("TEST SECTION A")

    # motor_go(clockwise, steptype", steps, stepdelay, verbose, initdelay)
#    input("TEST: Press <Enter> to continue  Full 180 turn Test1")
#    mymotortest.motor_go(False, "Full" , 100, .05, False, .05)
#    time.sleep(1)
#    input("TEST: Press <Enter> to continue  full 180 clockwise Test2")
#    mymotortest.motor_go(True, "Full" , 100, .05, True, .05)
#    time.sleep(1)
#    input("TEST: Press <Enter> to continue  full 180 no verbose Test3")
#    mymotortest.motor_go(False, "Full" , 100, .05, False, .05)
#    time.sleep(1)
#    input("TEST: Press <Enter> to continue  timedelay Test4")
#    mymotortest.motor_go(True, "Full" , 10, 1, True, .05)
#    time.sleep(1)
#    input("TEST: Press <Enter> to continue  full initdelay Test5")
#    mymotortest.motor_go(True, "Full" , 90, .01, True, 10)
#    time.sleep(1)

    # ========================== section B =========================
#    print("TEST SECTION B")

    # motor_go(clockwise, steptype", steps, stepdelay, verbose, initdelay)
#    input("TEST: Press <Enter> to continue  half Test6")
#    mymotortest.motor_go(False, "Half" , 400, .005, True, .05)
#    time.sleep(1)

#    input("TEST: Press <Enter> to continue 1/ 4 Test7")
#    mymotortest.motor_go(False, "1/4" , 800, .005, True, .05)
#    time.sleep(1)
#    input("TEST: Press <Enter> to continue 1/8 Test8")
#    mymotortest.motor_go(False, "1/8" , 1600, .005, True, .05)
#    time.sleep(1)

#    input("TEST: Press <Enter> to continue  1/16 Test9")
#    mymotortest.motor_go(False, "1/16" , 3200, .0001, True, .001)
#    time.sleep(1)
    mymotortest.motor_go(True, "1/16" , 3200, .0001, True, .001)

# Comment in for testing motor stop ,  put push button to VCC on GPIO 17
def button_callback(channel):
    print("Test file: Stopping motor")
    mymotortest.motor_stop()

# ===================MAIN===============================

if __name__ == '__main__':

    print("TEST START")
    main()
    GPIO.cleanup()
    print("TEST END")
    exit()

# =====================END===============================`

gavinlyonsrepo, 2024-01-29:
Hi 

The problem is the 'mymotortest' object is declared in 'main' so the  
button_callback cannot see it. So we need to ::
1. move the declaration of 'mymotortest' 
2. above line def main():
3.  below line 'GPIO.setup(4, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)'

```
    # ====== Tests for motor ======

    # Microstep Resolution MS1-MS3 -> GPIO Pin , can be set to (-1,-1,-1) to turn off
    GPIO_pins = (14, 15, 18)
    direction= 21       # Direction -> GPIO Pin
    step = 20      # Step -> GPIO Pin

    # Declare an named instance of class pass GPIO-PINs
    # (self, direction_pin, step_pin, mode_pins , motor_type):
    mymotortest = RpiMotorLib.A4988Nema(direction, step, GPIO_pins, "A4988")

```

regards

umsmin, 2024-01-29:
This was the first thing I try but then I gen another error.

`  File "/usr/src/RpiMotorLib-3.2/test/./A4988_Nema_Test.py", line 26
    mymotortest = RpiMotorLib.A4988Nema(direction, step, GPIO_pins, "A4988")
IndentationError: unexpected indent`

gavinlyonsrepo, 2024-01-29:
IndentationError: unexpected  indent 

Indentation refers to the spaces at the beginning of a code line.
Where in other programming languages the indentation in code is for readability only, the indentation in Python is very important.
You got to correct the indentation on line 26 remove the excess spaces on left hand side of line.

umsmin, 2024-01-29:
Thanks Python is not my friend I think. 

Now the errors gone. How can I change the Button from cative HIGH to active LOW?

gavinlyonsrepo, 2024-01-29:
> Thanks Python is not my friend I think.
> 
> Now the errors gone. How can I change the Button from cative HIGH to active LOW?

GPIO.setup(channel, GPIO.IN, pull_up_down=GPIO.PUD_UP)

  https://sourceforge.net/p/raspberry-gpio-python/wiki/Inputs/

umsmin, 2024-01-29:
Unfortunately he say 

`./A4988_Nema_Test.py
TEST START
Test file: Stopping motor
Test file: Stopping motor
Stop Motor Interrupt : RpiMotorLib:
TEST END`

umsmin, 2024-01-29:
I am checking also about the pin is working.
On commandline

```
root@pimicroscop:/usr/src/RpiMotorLib-3.2/test# raspi-gpio get 4
GPIO 4: level=1 fsel=0 func=INPUT
root@pimicroscop:/usr/src/RpiMotorLib-3.2/test# raspi-gpio get 4
GPIO 4: level=0 fsel=0 func=INPUT
```
The first without pushing and the second with

umsmin, 2024-01-29:
I changed the line 

`GPIO.add_event_detect(4, GPIO.RISING, callback=button_callback)`

to

`GPIO.add_event_detect(4, GPIO.FALLING, callback=button_callback)`

And the line

`GPIO.setup(channel, GPIO.IN, pull_up_down=GPIO.PUD_UP)`

to

`GPIO.setup(channel, GPIO.IN)`

because there is a hardware resistor.

Nothing happend

gavinlyonsrepo, 2024-02-01:
Hi 

What is happening?
What are you excepting and want  to happen?
Post your entire test file after all your  changes.
Show your wiring diagram.

regards

umsmin, 2024-02-01:
Hi,

I tried a lot of things but my endstop don't work in Python3. He work perfect in the command shell. So the wiring ist correct. You can read it a few post before. 

For now I use another tool with your library but without the endstop. 

Here the code who doesn't work:

`#!/usr/bin/env python3
""" test example file for rpiMotorlib.py A4988 NEMA"""

import time
import RPi.GPIO as GPIO

"""
# Next 3 lines for development, local library testing import
# Comment out in production release and change RpiMotorLib.A4988Nema to A4988Nema
import sys
sys.path.insert(0, '/home/pi/Documents/tech/RpiMotorLib/RpiMotorLib')
from RpiMotorLib import A4988Nema
"""

# Production installed library import
from RpiMotorLib import RpiMotorLib

channel = 17

# Comment in To Test motor stop, put a push button to VCC on GPIO 4
GPIO.setwarnings(False)
GPIO.setmode(GPIO.BCM)
#GPIO.setup(channel, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)
GPIO.setup(channel, GPIO.IN)

    # ====== Motor Init ======

# Microstep Resolution MS1-MS3 -> GPIO Pin , can be set to (-1,-1,-1) to turn off
GPIO_pins = (14, 15, 18)
direction= 21  # Direction -> GPIO Pin
step = 20      # Step -> GPIO Pin

# Declare an named instance of class pass GPIO-PINs
# (self, direction_pin, step_pin, mode_pins , motor_type):
mymotortest = RpiMotorLib.A4988Nema(direction, step, GPIO_pins, "A4988")

# Comment in for testing motor stop ,  put push button to VCC on GPIO 4
def button_callback(channel):
    print("Test file: Stopping motor")
    mymotortest.motor_stop()

def main():
    """main function loop"""

    # Comment in To Test motor stop , put push button to VCC on GPIO 4
    GPIO.add_event_detect(channel, GPIO.FALLING, callback=button_callback, bouncetime=200)

    mymotortest.motor_go(False, "1/16" , 3200, .0001, True, .01) # Hoch

#    mymotortest.motor_go(True, "1/16" , 3200, .0001, True, .01) # Runter

# ===================MAIN===============================

if __name__ == '__main__':

    print("Pimikroskop START")
    main()
    GPIO.cleanup()
    print("Pimikroskop ENDE")
    exit()

# =====================END===============================`

umsmin, 2024-02-01:
I wrote a small script and the endstop works.
The difference is that I used the gpiozero library for this.

`#!/usr/bin/env python3
`
from gpiozero import Button
taster = Button(17)
while True:
    if taster.is_pressed:
        print("Taster ist gedrückt!")
    else:
        print("Taster ist nicht gedrückt!")

gavinlyonsrepo, 2024-02-02:
Hi 

1.

If you using a push up button to Gnd with the  internal pull resistor  
the commented out line 
#GPIO.setup(channel, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)
needs to be changed to 
GPIO.setup(channel, GPIO.IN, pull_up_down=GPIO.PUD_UP)

2. 

If you using a push up button to Gnd without the  internal pull resistor and  using External resistor. code works below I have increased stepdelay to give more time to press button

[image]
```

#!/usr/bin/env python3
""" test example file for rpiMotorlib.py A4988 NEMA"""

import time
import RPi.GPIO as GPIO

# Production installed library import
from RpiMotorLib import RpiMotorLib

channel = 17
# Comment in To Test motor stop, put a push button to VCC on GPIO 4
GPIO.setwarnings(False)
GPIO.setmode(GPIO.BCM)
# Works with internal pullup resistor to VCC
#GPIO.setup(channel, GPIO.IN, pull_up_down=GPIO.PUD_UP)
# Works with external pull up resistor to VCC
GPIO.setup(channel, GPIO.IN)

    # ====== Motor Init ======

# Microstep Resolution MS1-MS3 -> GPIO Pin , can be set to (-1,-1,-1) to turn off
GPIO_pins = (14, 15, 18)
direction= 21  # Direction -> GPIO Pin
step = 20      # Step -> GPIO Pin

# Declare an named instance of class pass GPIO-PINs
# (self, direction_pin, step_pin, mode_pins , motor_type):
mymotortest = RpiMotorLib.A4988Nema(direction, step, GPIO_pins, "A4988")

# Comment in for testing motor stop ,  put push button to VCC on GPIO 4
def button_callback(channel):
    print("Test file: Stopping motor")
    mymotortest.motor_stop()

def main():
    """main function loop"""
    # Comment in To Test motor stop , put push button to VCC on GPIO 4
    GPIO.add_event_detect(channel, GPIO.FALLING, callback=button_callback, bouncetime=200)
    mymotortest.motor_go(False, "1/16" , 3200, .01, True, .01) # Hoch
#    mymotortest.motor_go(True, "1/16" , 3200, .0001, True, .01) # Runter

# ===================MAIN===============================

if __name__ == '__main__':

    print("Pimikroskop START")
    main()
    GPIO.cleanup()
    print("Pimikroskop ENDE")
    exit()

# =====================END===============================
```

umsmin, 2024-02-02:
Hi,
The schematic is that I have. I don't use internal Resistors because it't an endstop from a 3D-Printer. 
Is this code working on your Raspberry Pi? For me not. I get only the same than before. No movement.

```
Test file: Stopping motor
Stop Motor Interrupt : RpiMotorLib:
```

gavinlyonsrepo, 2024-02-02:
Hi 

Yes it works here, all modes internal and external resistor ,  Active high and active low.
I get the failure mode you describe if resistor is missing in external resistor mode.
Is is possibly the pull up external resistor is detective or not making proper contact?
Can you measure resistor and contact to circuit?
What value of resistor do you use?

Regards

umsmin, 2024-02-02:
That's funny. The Resistor is a 10k and well soldered SMD. The switch works also well.

Wich Raspberry do you use? I have the 3B
The OS is : Debian 11.8 with Kernel 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64
Python Version is : 3.9

gavinlyonsrepo, 2024-02-04:
Hi 
I am using 
    Raspberry PI 3 model b
    Raspbian , Debian 12 bookworm OS, , 64 bit.
    kernel : aarch64 Linux 6.1.0-rpi7-rpi-v8
    RPI.GPIO version 0.7.1a4
   python 3.11.2

Maybe its a grounding issue , Are all grounds in circuit connected together?

umsmin, 2024-02-05:
Let me try to upgrade Python and also Debian to 12. Maybe something wrong there? I don't know but that's the difference. 
About the grounding, the wires running directly to the ground on a small board who I made for the connection between the Pi the A4988, capacitor and the connectors for Stepper Motor-Power and the Endswitch.

gavinlyonsrepo, 2024-03-02:
Hi 

Any update?

umsmin, 2024-03-03:
Unfortunately no. I now use a web interface and no endstop.

gavinlyonsrepo, 2024-03-04:
Hi

The first  problem where the function in example file was out of scope, when commented back in,
was corrected in commit number  4ba2fd6. 
The second problem is most likely a user hardware issue in my view. 
No fault found with rpiMotorLib or RPi.GPIO. 
Closing issue. 

regards

closed 2024-03-04 by gavinlyonsrepo