gavinlyonsrepo/RpiMotorLib #26: Rpi5

lankycanadian, 2024-03-27:
I'm getting an error "cannot determine soc peripheral base address" with the GPIO library.  It looks as though GPIO has been replaced on the rpi5 
https://www.tomshardware.com/how-to/control-raspberry-pi-5-gpio-with-python-3

Do you have any plans to update this library to work on a Pi 5?

bmagsalan, 2024-04-18:
> I'm getting an error "cannot determine soc peripheral base address" with the GPIO library. It looks as though GPIO has been replaced on the rpi5 https://www.tomshardware.com/how-to/control-raspberry-pi-5-gpio-with-python-3
> 
> Do you have any plans to update this library to work on a Pi 5?

Yeah I ran into the same issue so I wrote a shared library for it using c++ and wiringPi.

Tested with arm64 bookworm on Raspberry Pi 5 with 2 x 28BYJ-48  stepper motors.

https://github.com/bmagsalan/StepperMotor

Motors movement is asynchronous so you can move them at the same time.

The library is scalable so feel free to expand on it.

Julz124, 2025-02-01:
> Yeah I ran into the same issue so I wrote a shared library for it using c++ and wiringPi.
>
> Tested with arm64 bookworm on Raspberry Pi 5 with 2 x 28BYJ-48 stepper motors. [...]

Real cool guide and library you wrote there, but I have one question.
The A4988 driver got a direction and step pin (which both can be low -> ground pin), a sleep and reset (also "low" and both bridged to one another) and at last the microstep resolution selection pins (MS1, MS2 and MS3) but your StepperMotor-class got 4 parameters to initialize it's pins.
The question is, which pins of the A4988 driver should I use for which location?